thankthemaker/rESCue #4: piezo buzzer

Koxx3, 2021-05-25:
Hello,
very nice project !
I am curious about the piezo buzzer.
it works with an audible noise even in the enclosure with 3.3V ?
did you measure the current ? not drawing to much current from ESP pin ?
Regards,
F.

thankthemaker, 2021-05-25:
Hi,
to get a really loud sound, you'll need to add a simple transistor, e.g. a BC547.
If a buzzer does any harm to your ESP depends on the type. Piezo buzzer normally don't drain more than 20mA which is ok for the ESP32. Magnetic buzzer may drain much more and could damage the MCU.
Here is a good description:
https://diyi0t.com/active-passive-buzzer-arduino-esp8266-esp32/

I'd always recommend to use a transistor.

closed 2021-05-25 by thankthemaker

Koxx3, 2021-05-25:
thanks for the answer ;)
20 mA is already a lot for the ESP32 (12mA recommended if I have good memory)  ;)
I also think transistor is a good idea :D

thankthemaker, 2021-05-25:
According to the table on the page I posted, 12mA is for ESP8266.
The ESP32 can handle 20mA. 
At least I never damaged an ESP32 by connecting a piezo buzzer.