thankthemaker/rESCue #5: schematics for PCB

Koxx3, 2021-05-25:
hello,
could you publish the PCB schematics on the github ?
Regards,
F.

thankthemaker, 2021-05-25:
I'm not the creator of the PCB, so I can't publish the schematics.
Jonas is the guy who did a amazing job and he also told me he'll publish the complete project as open hardware once it is finished.

Koxx3, 2021-05-25:
awesome ;)
thanks for the feedback.
I am creating something a bit similar for escooters (with multiple controllers capability) ;)
https://github.com/Koxx3/SmartController_SmartDisplay_ESP32

If you have Telegram, we can talk (https://t.me/koxx3) 👍

thankthemaker, 2021-05-25:
Sounds great. Have to read it.

thankthemaker, 2021-06-05:
This repository contains only the firmware.
The hardware, once released as open hardware, will be placed in a new repository.

closed 2021-06-05 by thankthemaker