thankthemaker/rESCue #12: Understanding BLE characteristics

Koxx3, 2021-06-28:
Hello,

I am trying to understand how VESC Tools use the 2 BLE characteristics for Serial... 

```
21:35:47.338 > BLH - BLE connecting
21:35:47.828 > BLH - onAuthenticationComplete desc : success
21:35:47.830 > Client ID: 0, Address: cc:21:19:9e:7f:c8, Subvalue 1, Characteristics beb5483e-36e1-4688-b7f5-ea07361b26a0
21:35:47.832 > Client ID: 0, Address: cc:21:19:9e:7f:c8, Subvalue 1, Characteristics beb5483e-36e1-4688-b7f5-ea07361b26a6
21:35:47.833 > Client ID: 0, Address: cc:21:19:9e:7f:c8, Subvalue 1, Characteristics beb5483e-36e1-4688-b7f5-ea07361b26af
21:35:47.834 > Client ID: 0, Address: cc:21:19:9e:7f:c8, Subvalue 1, Characteristics beb5483e-36e1-4688-b7f5-ea07361b26a5
21:35:47.835 > Client ID: 0, Address: cc:21:19:9e:7f:c8, Subvalue 1, Characteristics 6e400003-b5a3-f393-e0a9-e50e24dcca9e
21:35:47.991 > BLH - onRead : 6e400003-b5a3-f393-e0a9-e50e24dcca9e
21:35:47.993 > BLH - Read VESC_CHARACTERISTIC_UUID_TX_UUID
21:35:47.993 > onRead to characteristics: 6e400003-b5a3-f393-e0a9-e50e24dcca9e
21:35:47.993 > onRead - value:  -  - length: 0
21:35:47.994 >
21:35:48.186 > BLH - onRead : 6e400002-b5a3-f393-e0a9-e50e24dcca9e
21:35:48.187 > BLH - Read VESC_CHARACTERISTIC_UUID_RX_UUID
21:35:48.188 > onRead to characteristics: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
21:35:48.189 > onRead - value:  -  - length: 0
21:35:48.189 >
```

On my side, it's weird because it tries to read datas.

But from my understanding, it should first send a write to push serial to VESC (write with VESC_CHARACTERISTIC_UUID_TX_UUID) in order to get an answer (then read with VESC_CHARACTERISTIC_UUID_RX_UUID).

Could you tell me what am I missing  please ?

Regards,
F.

Koxx3, 2021-06-28:
a bit better .... but still failing to read firmware version ... 

```

21:51:28.326 > BLH - BLE connecting
21:51:28.789 > BLH - onAuthenticationComplete desc : success
21:51:28.792 > Client ID: 0, Address: cc:21:19:9e:7f:c8, Subvalue 1, Characteristics beb5483e-36e1-4688-b7f5-ea07361b26a0
21:51:28.792 > Client ID: 0, Address: cc:21:19:9e:7f:c8, Subvalue 1, Characteristics beb5483e-36e1-4688-b7f5-ea07361b26a6
21:51:28.794 > Client ID: 0, Address: cc:21:19:9e:7f:c8, Subvalue 1, Characteristics beb5483e-36e1-4688-b7f5-ea07361b26af
21:51:28.795 > Client ID: 0, Address: cc:21:19:9e:7f:c8, Subvalue 1, Characteristics beb5483e-36e1-4688-b7f5-ea07361b26a5
21:51:28.796 > Client ID: 0, Address: cc:21:19:9e:7f:c8, Subvalue 1, Characteristics 6e400003-b5a3-f393-e0a9-e50e24dcca9e
21:51:28.955 > BLH - onRead : 6e400003-b5a3-f393-e0a9-e50e24dcca9e
21:51:28.957 > BLH - Read VESC_CHARACTERISTIC_UUID_TX_UUID
21:51:28.957 > onRead to characteristics: 6e400003-b5a3-f393-e0a9-e50e24dcca9e
21:51:28.958 > onRead - value:  -  - length: 0
21:51:28.958 >
21:51:29.021 > BLH - onRead : 6e400002-b5a3-f393-e0a9-e50e24dcca9e
21:51:29.023 > BLH - Read VESC_CHARACTERISTIC_UUID_RX_UUID
21:51:29.024 > onRead to characteristics: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
21:51:29.024 > onRead - value:  -  - length: 0
21:51:29.025 >
21:51:29.134 > BLH - Write : VESC_CHARACTERISTIC_UUID_RX_UUID
21:51:29.134 > onWrite to characteristics: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
21:51:29.136 > onWrite - length: 6
21:51:29.136 >
21:51:29.136 > sendPayload - Package to send: 02 01 00 00 00 03 00
21:51:29.219 > End of message reached!
21:51:29.220 > SRC received: 63948
21:51:29.220 > SRC calc: 63948
21:51:29.220 > Received: 02 17 00 05 02 34 31 30 00 3e 00 1e 00 05 47 39 32 39 34 39 36 00 00 00 00 f9 cc 03 00
21:51:29.221 >
21:51:29.221 > Payload :      00 05 02 34 31 30 00 3e 00 1e 00 05 47 39 32 39 34 39 36 00 00 00 00
21:51:29.222 >
21:51:29.222 > Package received: 02 17 00 05 02 34 31 30 00 3e 00 1e 00 05 47 39 32 39 34 39 36 00 00 00 00 f9 cc 03 00
21:51:29.223 > BLE from VESC:
21:51:29.475 > BLH - Write : VESC_CHARACTERISTIC_UUID_RX_UUID
21:51:29.476 > onWrite to characteristics: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
21:51:29.476 > onWrite - length: 6
21:51:29.476 >
21:51:29.476 > sendPayload - Package to send: 02 01 0e e1 ce 03 00
21:51:29.478 > BLH - Write : VESC_CHARACTERISTIC_UUID_RX_UUID
21:51:29.478 > onWrite to characteristics: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
21:51:29.478 > onWrite - length: 6
21:51:29.479 >
21:51:29.479 > sendPayload - Package to send: 02 01 11 02 10 03 00
21:51:30.205 > BLH - Write : VESC_CHARACTERISTIC_UUID_RX_UUID
21:51:30.207 > onWrite to characteristics: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
21:51:30.207 > onWrite - length: 6
21:51:30.208 >
21:51:30.208 > sendPayload - Package to send: 02 01 00 00 00 03 00
21:51:30.449 > BLH - Write : VESC_CHARACTERISTIC_UUID_RX_UUID
21:51:30.451 > onWrite to characteristics: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
21:51:30.451 > onWrite - length: 6
21:51:30.451 >
21:51:30.451 > sendPayload - Package to send: 02 01 0e e1 ce 03 00
21:51:30.452 > BLH - Write : VESC_CHARACTERISTIC_UUID_RX_UUID
21:51:30.453 > onWrite to characteristics: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
21:51:30.454 > onWrite - length: 6
21:51:30.454 >
21:51:30.454 > sendPayload - Package to send: 02 01 11 02 10 03 00
21:51:31.922 > wifi portal open !
21:51:32.108 > Client ID: 0, Address: cc:21:19:9e:7f:c8, Subvalue 0, Characteristics beb5483e-36e1-4688-b7f5-ea07361b26a0 
21:51:32.110 > Client ID: 0, Address: cc:21:19:9e:7f:c8, Subvalue 0, Characteristics beb5483e-36e1-4688-b7f5-ea07361b26a6
21:51:32.111 > Client ID: 0, Address: cc:21:19:9e:7f:c8, Subvalue 0, Characteristics beb5483e-36e1-4688-b7f5-ea07361b26af
21:51:32.111 > Client ID: 0, Address: cc:21:19:9e:7f:c8, Subvalue 0, Characteristics beb5483e-36e1-4688-b7f5-ea07361b26a5
21:51:32.112 > Client ID: 0, Address: cc:21:19:9e:7f:c8, Subvalue 0, Characteristics 6e400003-b5a3-f393-e0a9-e50e24dcca9e
21:51:32.115 > BLH - BLE disconnected
```

thankthemaker, 2021-06-28:
The message looks wrong
`02 01 00 00 00 03 00`

Should be 
`02 01 00 00 00 03`

I recommend to get something like BlueSee and try to send from it.

Koxx3, 2021-06-29:
thanks a lot for your answer.
ok, thanks, it's fixed.

Why is Vesc Tools try to read the 2 characteristics at the begining (with 0 bytes payload) ?
And sometime, it doesn't continue with a "Write : VESC_CHARACTERISTIC_UUID_RX_UUID" ... and stay blocked.

Koxx3, 2021-06-29:
example of dead lock situation :

```
19:26:26.290 > BLH - BLE connecting
19:26:26.746 > BLH - onAuthenticationComplete desc : success
19:26:26.748 > Client ID: 0, Address: cc:21:19:9e:7f:c8, Subvalue 1, Characteristics beb5483e-36e1-4688-b7f5-ea07361b26fe
19:26:26.749 > Client ID: 0, Address: cc:21:19:9e:7f:c8, Subvalue 1, Characteristics beb5483e-36e1-4688-b7f5-ea07361b26ff
19:26:26.750 > Client ID: 0, Address: cc:21:19:9e:7f:c8, Subvalue 1, Characteristics beb5483e-36e1-4688-b7f5-ea07361b26a0
19:26:26.752 > Client ID: 0, Address: cc:21:19:9e:7f:c8, Subvalue 1, Characteristics beb5483e-36e1-4688-b7f5-ea07361b26a6
19:26:26.753 > Client ID: 0, Address: cc:21:19:9e:7f:c8, Subvalue 1, Characteristics beb5483e-36e1-4688-b7f5-ea07361b26af
19:26:26.753 > Client ID: 0, Address: cc:21:19:9e:7f:c8, Subvalue 1, Characteristics beb5483e-36e1-4688-b7f5-ea07361b26a5
19:26:27.092 > BLH - onRead : 6e400003-b5a3-f393-e0a9-e50e24dcca9e
19:26:27.094 > BLH - Read VESC_CHARACTERISTIC_UUID_TX_UUID
19:26:27.094 > onRead - value:  -  - length: 0
19:26:27.678 > BLH - Read VESC_CHARACTERISTIC_UUID_RX_UUID
19:26:27.678 > onRead - value:  -  - length: 0
19:26:27.678 >
19:26:27.870 > Client ID: 0, Address: cc:21:19:9e:7f:c8, Subvalue 1, Characteristics 6e400003-b5a3-f393-e0a9-e50e24dcca9e 
```
... the nothing... 
I really don't understand. sometimes, it works, sometimes not.

thankthemaker, 2021-06-30:
I never experienced such a behaviour. rESCue uses UART only as a transparent proxy, therefore every byte is just send exactly as is in both directions. rESCue doesn't build or evaluate any messages between VESC and VESC tool app.
When using CANBUS, rESCue has to translate the messages, but also with CAN I never experienced a deadlock.

closed 2021-06-30 by thankthemaker

Koxx3, 2021-07-01:
ok, thanks