thankthemaker/rESCue #16: CAN bus / BLE gateway

Koxx3, 2021-11-12:
Hello,
do you think it would be possible to make a CAN <> BLE gateway ?

thankthemaker, 2021-11-13:
Hi, that's exactly what rESCue already does, at least if you're talking of the VESC protocol. 
If you're talking about a generic solution, sure, this will also be possible but isn't part of rESCue already.

Could you please tell me a bit more, what exactly your intention is all about?

Koxx3, 2021-11-13:
I'd like to use it as a simple gateway for either BLE <> UART or BLE <> CAN.

BLE <> UART works fine. Now, I try the CAN version.

VESC configuration is in CAN_STATUS_1_2_3_4_5 enabled at 50Hz and 500KBps.
I currently have a single VESC (but I want it to work on dual CAN VESC).

I understood the proxyIn/proxyOut and got some results.
First, I had to reduce Logger traces. Verbose seems to give some issues (certainly slowing down the CAN frame processing).

But once connected in BLE, I can read real time datas, read MC/APP conf, but I can't write APP conf. Any idea why ? It's a long packet (but like MC conf), but I don't understand why this one is not working.

My test fork is here (I change the pinout for UART & CAN) :
https://github.com/Koxx3/rESCue

Koxx3, 2021-11-13:
I've just push the commit in the fork ;)

thankthemaker, 2021-11-13:
OK, so we're talking of the VESC protocol.
 I'll have to have a look at it. It might be a problem with the length of the config. Normally all data is passed to the CAN bus as is and vice versa, but due to different framing (package sizes) it has to be repacked. The VESC protocol distinguishes between different package sizes and I guess it could be a problem if e.g. the last CAN package isn't a complete package.

Koxx3, 2021-11-13:
even for MC conf writing, it doesn't work all the time

Koxx3, 2021-11-13:
working sequence : 

`
FWC: [NOTICE] : BleServer BLE to VESC : len = 20
FWC: [NOTICE] : CanBus Proxy in  : STREAM => CAN, command 13, length 458, longPacket 1
FWC: [NOTICE] : BleServer BLE to VESC : len = 20
FWC: [NOTICE] : BleServer BLE to VESC : len = 20
FWC: [NOTICE] : BleServer BLE to VESC : len = 20
FWC: [NOTICE] : BleServer BLE to VESC : len = 20
FWC: [NOTICE] : BleServer BLE to VESC : len = 20
FWC: [NOTICE] : BleServer BLE to VESC : len = 20
FWC: [NOTICE] : BleServer BLE to VESC : len = 20
FWC: [NOTICE] : BleServer BLE to VESC : len = 20
FWC: [NOTICE] : BleServer BLE to VESC : len = 20
FWC: [NOTICE] : BleServer BLE to VESC : len = 20
FWC: [NOTICE] : BleServer BLE to VESC : len = 20
FWC: [NOTICE] : BleServer BLE to VESC : len = 20
FWC: [NOTICE] : BleServer BLE to VESC : len = 20
FWC: [NOTICE] : BleServer BLE to VESC : len = 20
FWC: [NOTICE] : BleServer BLE to VESC : len = 20
FWC: [NOTICE] : BleServer BLE to VESC : len = 20
FWC: [NOTICE] : BleServer BLE to VESC : len = 20
FWC: [NOTICE] : BleServer BLE to VESC : len = 20
FWC: [NOTICE] : BleServer BLE to VESC : len = 20
FWC: [NOTICE] : BleServer BLE to VESC : len = 20
FWC: [NOTICE] : BleServer BLE to VESC : len = 20
FWC: [NOTICE] : BleServer BLE to VESC : len = 20
FWC: [NOTICE] : BleServer BLE to VESC : len = 4
FWC: [NOTICE] : BleServer BLE to VESC : len = 6
FWC: [NOTICE] : CanBus Proxy in  : STREAM => CAN, command 14, length 1, longPacket 0
FWC: [NOTICE] : CanBus Proxy out : CAN => STREAM, sending 2 bytes
FWC: [NOTICE] : CanBus Proxy out : CAN => STREAM, sending 458 bytes
FWC: [NOTICE] : BleServer VESC to BLE : len = 471`

thankthemaker, 2021-11-13:
Are you using the VESC tool app for testing?

Koxx3, 2021-11-13:
yes, android vesctool with samsung Galaxy S10E.

Koxx3, 2021-11-13:
failing writing APP conf :
 
> FWC: [NOTICE] : BleServer BLE to VESC : len = 20
> FWC: [NOTICE] : CanBus Proxy in  : STREAM => CAN, command 16, length 419, longPacket 1
> FWC: [NOTICE] : BleServer BLE to VESC : len = 20
> FWC: [NOTICE] : BleServer BLE to VESC : len = 20
> FWC: [NOTICE] : BleServer BLE to VESC : len = 20
> FWC: [NOTICE] : BleServer BLE to VESC : len = 20
> FWC: [NOTICE] : BleServer BLE to VESC : len = 20
> FWC: [NOTICE] : BleServer BLE to VESC : len = 20
> FWC: [NOTICE] : BleServer BLE to VESC : len = 20
> FWC: [NOTICE] : BleServer BLE to VESC : len = 20
> FWC: [NOTICE] : BleServer BLE to VESC : len = 20
> FWC: [NOTICE] : BleServer BLE to VESC : len = 20
> FWC: [NOTICE] : BleServer BLE to VESC : len = 20
> FWC: [NOTICE] : BleServer BLE to VESC : len = 20
> FWC: [NOTICE] : BleServer BLE to VESC : len = 20
> FWC: [NOTICE] : BleServer BLE to VESC : len = 20
> FWC: [NOTICE] : BleServer BLE to VESC : len = 20
> FWC: [NOTICE] : BleServer BLE to VESC : len = 20
> FWC: [NOTICE] : BleServer BLE to VESC : len = 20
> FWC: [NOTICE] : BleServer BLE to VESC : len = 20
> FWC: [NOTICE] : BleServer BLE to VESC : len = 20
> FWC: [NOTICE] : BleServer BLE to VESC : len = 20
> FWC: [NOTICE] : BleServer BLE to VESC : len = 5
> FWC: [NOTICE] : CanBus Proxy out : CAN => STREAM, sending 2 bytes
> FWC: [NOTICE] : BleServer VESC to BLE : len = 7

thankthemaker, 2021-11-13:
Ok, seems your fork is also behind the original master. Did you also try the latest master version from my original repository?
There were some changes, e.g. I separated the buffers to avoid influences of proxy / non proxy packages.

Koxx3, 2021-11-13:
oh ? I missed something then.
I think I've just checked out this one this morning.
Can you point me to the right link please ?

thankthemaker, 2021-11-13:
I'm referring to what the comparison of my and your master branches shows me:

https://github.com/thankthemaker/rESCue/compare/master...Koxx3:master

So, either you have to update your fork or check out the master of the original repo.

Koxx3, 2021-11-13:
ok, I'll restart from the start ;)
I'll keep you updated.
thanks

Koxx3, 2021-11-13:
updated !
and still the same issue.

Koxx3, 2021-11-13:
in fact, it's weird.
when I use the direct VESC BLE, I have the small green toast at the bottom confirming the write operation in Android vesctool.
when I use rescue, I don't have this toast, but datas seems to be taken into account by the VESC (may be partially ?)

Koxx3, 2021-11-13:
[image]

Koxx3, 2022-01-18:
I managed to get it working ;)

closed 2022-01-18 by Koxx3