thankthemaker/rESCue #26: CAN / BLE gateway with new officiel VESC 3.01 android app

Koxx3, 2022-01-18:
Hello,
it worked before, but now, it can't detect other VESC in CAN bus.
did you try the latest officiel android VESCTool release ?
Regards,
F.

Koxx3, 2022-01-18:
[image]

Koxx3, 2022-01-18:
#
[image]

Koxx3, 2022-01-18:
First, it should show the second VESC
Second, tje timeout warning.

thankthemaker, 2022-01-20:
I haven't tried yet. Thanks for the hint, will have a look at it.

Koxx3, 2022-01-21:
```
// BLE<>UART sequence
10:29:38.325 > BluetoothHandler BLE => VESC : len = 006 / 02 01 3e d7 9d 03  // request CAN PING
10:29:40.903 > BluetoothHandler VESC => BLE : len = 006 / 02 02 3e c8 7e de 
10:29:40.927 > BluetoothHandler VESC => BLE : len = 001 / 03 
10:29:40.980 > BluetoothHandler BLE => VESC : len = 006 / 02 01 00 00 00 03 
10:29:40.982 > BluetoothHandler VESC => BLE : len = 002 / 02 25
10:29:41.007 > BluetoothHandler VESC => BLE : len = 040 / 00 05 02 55 42 4f 58 5f 37 35 5f 33 30 30 5f 54 59 50 45 43 
10:29:41.028 > BluetoothHandler VESC => BLE : len = 020 / 00 20 00 23 00 05 50 33 53 30 37 33 20 00 00 00 00 51 72 03 
10:29:41.047 > BluetoothHandler BLE => VESC : len = 008 / 02 03 22 c8 00 77 5b 03 
10:29:41.051 > BluetoothHandler VESC => BLE : len = 015 / 02 25 00 05 02 55 42 4f 58 5f 37 35 5f 33 30
10:29:41.075 > BluetoothHandler VESC => BLE : len = 027 / 30 5f 54 59 50 45 43 00 4e 00 37 00 0e 50 50 59 58 38 31 20 
10:29:41.097 > BluetoothHandler VESC => BLE : len = 007 / 00 00 00 00 e8 9e 03 

// BLE<>CAN sequence
10:33:45.187 > BluetoothHandler BLE => VESC : len = 006 / 02 01 3e d7 9d 03  // request CAN PING 
10:33:47.766 > BluetoothHandler VESC => BLE : len = 008 / 02 03 01 3e c8 70 20 03
10:33:50.260 > BluetoothHandler BLE => VESC : len = 006 / 02 01 00 00 00 03 
10:33:50.276 > BluetoothHandler VESC => BLE : len = 042 / 02 25 00 05 02 55 42 4f 58 5f 37 35 5f 33 30 30 5f 54 59 50
10:33:50.307 > BluetoothHandler VESC => BLE : len = 022 / 45 43 00 20 00 23 00 05 50 33 53 30 37 33 20 00 00 00 00 51 
10:33:50.338 > BluetoothHandler VESC => BLE : len = 002 / 72 03 
```

Koxx3, 2022-01-21:
in fact, even with older Vesctool, it didn't work.
Archive available here, tried 2.0.0 :
https://drive.google.com/drive/folders/1B7TBKsgpeKyqS-BKL1QFh0fiHb8Xv3XI

thankthemaker, 2022-01-21:
Please try the latest version 2.0.4
There was an issue of one of the older versions.

Koxx3, 2022-01-21:
i don't have the apk.
but not working with 3.0.1 anyway. so, not good since it's the only version
available on play store 😉
i found a bug in proxyin on short packet processing. i'll explain later.
wizzard can't work with this bug.

thankthemaker, 2022-01-22:
I was talking about version 2.0.4 of rESCue, just to be precise. I‘ll do some tests with VESCtool later, but I know I tested the 2.0.4 of rESCue wit a 3.0.0 beta of VESCtool several times.

thankthemaker, 2022-04-16:
Should work starting with version 2.1.3

closed 2022-04-16 by thankthemaker

Koxx3, 2022-04-16:
thanks ! 
dit you test the can<>ble gw on fw 5.3 and 6.0 beta ?

thankthemaker, 2022-04-16:
I only tested it on 5.3.

Koxx3, 2022-04-18:
hello,
still having troubles 
My test conditions : 
- disable ledcontroller logs
- disabled CAN status 1/2/3/4/5 for less traces
- set verbose logs on rescue

BLE connects, but it fails try to get datas from CAN bus.

```
[E][Preferences.cpp:49] begin(): nvs_open failed: NOT_FOUND
[E][AppConfiguration.cpp:16] readPreferences(): no config file found
[E][AppConfiguration.cpp:23] readPreferences(): readPreferences:
FWC: [NOTICE] : BatteryMonitor initializing...
FWC: [NOTICE] : BleServer waiting a client connection to notify...
FWC: [NOTICE] : Ws28xxController initializing ...
FWC: [NOTICE] : Ws28xxController run startSequence
FWC: [NOTICE] : rESCue  sw-version 2.1.3 is happily running on hw-version 3.1
FWC: [NOTICE] : CanBus requestFirmwareVersion
FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485721 DLC 3, [26, 0, 0, 0, 0, 0, 0, 0]
FWC: [VERBOSE] : CanBus name=, dutycycle=0.000000, erpm=0.000000, current=0.000000, ampHours=0.000000, ampHoursCharged=0.000000, wattHours=0.000000, wattHoursCharged=0.000000, mosfetTemp=0.000000, motorTemp=0.000000, inputVoltage=0.000000, tachometer=0.000000, pidOutput=0.000000, pitch=0.000000, roll=0.000000, loopTime=0, motorCurrent=0.000000, motorPosition=0.000000, balanceState=0, switchState=0, adc1=0.000000, adc2=0.000000, fault=0
FWC: [NOTICE] : CanBus requestFirmwareVersion
FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485721 DLC 3, [26, 0, 0, 0, 0, 0, 0, 0]
FWC: [NOTICE] : CanBus requestFirmwareVersion
FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485721 DLC 3, [26, 0, 0, 0, 0, 0, 0, 0]
FWC: [VERBOSE] : CanBus name=, dutycycle=0.000000, erpm=0.000000, current=0.000000, ampHours=0.000000, ampHoursCharged=0.000000, wattHours=0.000000, wattHoursCharged=0.000000, mosfetTemp=0.000000, motorTemp=0.000000, inputVoltage=0.000000, tachometer=0.000000, pidOutput=0.000000, pitch=0.000000, roll=0.000000, loopTime=0, motorCurrent=0.000000, motorPosition=0.000000, balanceState=0, switchState=0, adc1=0.000000, adc2=0.000000, fault=0
FWC: [NOTICE] : CanBus requestFirmwareVersion
FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485721 DLC 3, [26, 0, 0, 0, 0, 0, 0, 0]
FWC: [NOTICE] : CanBus requestFirmwareVersion
FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485721 DLC 3, [26, 0, 0, 0, 0, 0, 0, 0]
FWC: [ERROR] CANBUS initialization failed
FWC: [VERBOSE] : CanBus name=, dutycycle=0.000000, erpm=0.000000, current=0.000000, ampHours=0.000000, ampHoursCharged=0.000000, wattHours=0.000000, wattHoursCharged=0.000000, mosfetTemp=0.000000, motorTemp=0.000000, inputVoltage=0.000000, tachometer=0.000000, pidOutput=0.000000, pitch=0.000000, roll=0.000000, loopTime=0, motorCurrent=0.000000, motorPosition=0.000000, balanceState=0, switchState=0, adc1=0.000000, adc2=0.000000, fault=0
FWC: [VERBOSE] : CanBus name=, dutycycle=0.000000, erpm=0.000000, current=0.000000, ampHours=0.000000, ampHoursCharged=0.000000, wattHours=0.000000, wattHoursCharged=0.000000, mosfetTemp=0.000000, motorTemp=0.000000, inputVoltage=0.000000, tachometer=0.000000, pidOutput=0.000000, pitch=0.000000, roll=0.000000, loopTime=0, motorCurrent=0.000000, motorPosition=0.000000, balanceState=0, switchState=0, adc1=0.000000, adc2=0.000000, fault=0
FWC: [VERBOSE] : CanBus name=, dutycycle=0.000000, erpm=0.000000, current=0.000000, ampHours=0.000000, ampHoursCharged=0.000000, wattHours=0.000000, wattHoursCharged=0.000000, mosfetTemp=0.000000, motorTemp=0.000000, inputVoltage=0.000000, tachometer=0.000000, pidOutput=0.000000, pitch=0.000000, roll=0.000000, loopTime=0, motorCurrent=0.000000, motorPosition=0.000000, balanceState=0, switchState=0, adc1=0.000000, adc2=0.000000, fault=0
FWC: [VERBOSE] : CanBus name=, dutycycle=0.000000, erpm=0.000000, current=0.000000, ampHours=0.000000, ampHoursCharged=0.000000, wattHours=0.000000, wattHoursCharged=0.000000, mosfetTemp=0.000000, motorTemp=0.000000, inputVoltage=0.000000, tachometer=0.000000, pidOutput=0.000000, pitch=0.000000, roll=0.000000, loopTime=0, motorCurrent=0.000000, motorPosition=0.000000, balanceState=0, switchState=0, adc1=0.000000, adc2=0.000000, fault=0
FWC: [VERBOSE] : CanBus name=, dutycycle=0.000000, erpm=0.000000, current=0.000000, ampHours=0.000000, ampHoursCharged=0.000000, wattHours=0.000000, wattHoursCharged=0.000000, mosfetTemp=0.000000, motorTemp=0.000000, inputVoltage=0.000000, tachometer=0.000000, pidOutput=0.000000, pitch=0.000000, roll=0.000000, loopTime=0, motorCurrent=0.000000, motorPosition=0.000000, balanceState=0, switchState=0, adc1=0.000000, adc2=0.000000, fault=0
FWC: [NOTICE] : BleServer Client connected: 56:63:d9:0b:47:77
FWC: [NOTICE] : BleServer Multi-connect support: start advertising
FWC: [VERBOSE] : CanBus name=, dutycycle=0.000000, erpm=0.000000, current=0.000000, ampHours=0.000000, ampHoursCharged=0.000000, wattHours=0.000000, wattHoursCharged=0.000000, mosfetTemp=0.000000, motorTemp=0.000000, inputVoltage=0.000000, tachometer=0.000000, pidOutput=0.000000, pitch=0.000000, roll=0.000000, loopTime=0, motorCurrent=0.000000, motorPosition=0.000000, balanceState=0, switchState=0, adc1=0.000000, adc2=0.000000, fault=0
FWC: [NOTICE] : BleServer Client ID: 0, Address: 56:63:d9:0b:47:77, Subvalue 1, Characteristics 6e400003-b5a3-f393-e0a9-e50e24dcca9e 
FWC: [NOTICE] : BleServer onWrite to characteristics: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
FWC: [NOTICE] : BleServer onWrite - value: ␂␁

FWC: [VERBOSE] : BleCanProxy Proxy in, command 0, length 1

FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485721 DLC 3, [27, 0, 0, 0, 0, 0, 0, 0]
[... cut ...] 0, switchState=0, adc1=0.000000, adc2=0.000000, fault=0
FWC: [VERBOSE] : CanBus name=, dutycycle=0.000000, erpm=0.000000, current=0.000000, ampHours=0.000000, ampHoursCharged=0.000000, wattHours=0.000000, wattHoursCharged=0.000000, mosfetTemp=0.000000, motorTemp=0.000000, inputVoltage=0.000000, tachometer=0.000000, pidOutput=0.000000, pitch=0.000000, roll=0.000000, loopTime=0, motorCurrent=0.000000, motorPosition=0.000000, balanceState=0, switchState=0, adc1=0.000000, adc2=0.000000, fault=0

```

Koxx3, 2022-04-18:
traces with only CAN STATUS 1 enabled : 

```
FWC: [NOTICE] : CanBus requestRealtimeData
FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485721 DLC 7, [26, 0, 50, 0, 0, 135, 195, 0]
FWC: [NOTICE] : CanBus requestBalanceData
FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485721 DLC 3, [26, 0, 79, 0, 0, 0, 0, 0]
#1 from 0x00000902, DLC 8, data [0      0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #1, type 
#2 from 0x00000902, DLC 8, data [0      0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #2, type
FWC: [NOTICE] : CanBus requestRealtimeData
FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485721 DLC 7, [26, 0, 50, 0, 0, 135, 195, 0]
FWC: [NOTICE] : CanBus requestBalanceData
FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485721 DLC 3, [26, 0, 79, 0, 0, 0, 0, 0]
#1 from 0x00000902, DLC 8, data [0      0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #1, type
#2 from 0x00000902, DLC 8, data [0      0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #2, type 
FWC: [NOTICE] : CanBus requestRealtimeData
FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485721 DLC 7, [26, 0, 50, 0, 0, 135, 195, 0]
FWC: [NOTICE] : CanBus requestBalanceData
FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485721 DLC 3, [26, 0, 79, 0, 0, 0, 0, 0]
#1 from 0x00000902, DLC 8, data [0      0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #1, type 
#2 from 0x00000902, DLC 8, data [0      0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #2, type
FWC: [NOTICE] : CanBus requestRealtimeData
FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485721 DLC 7, [26, 0, 50, 0, 0, 135, 195, 0]
FWC: [NOTICE] : CanBus requestBalanceData
FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485721 DLC 3, [26, 0, 79, 0, 0, 0, 0, 0]
#1 from 0x00000902, DLC 8, data [0      0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #1, type
#2 from 0x00000902, DLC 8, data [0      0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #2, type
FWC: [NOTICE] : CanBus requestRealtimeData
FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485721 DLC 7, [26, 0, 50, 0, 0, 135, 195, 0]
FWC: [NOTICE] : CanBus requestBalanceData
FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485721 DLC 3, [26, 0, 79, 0, 0, 0, 0, 0]
#1 from 0x00000902, DLC 8, data [0      0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #1, type
#2 from 0x00000902, DLC 8, data [0      0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #2, type
FWC: [NOTICE] : CanBus requestRealtimeData
FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485721 DLC 7, [26, 0, 50, 0, 0, 135, 195, 0]
FWC: [NOTICE] : CanBus requestBalanceData
FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485721 DLC 3, [26, 0, 79, 0, 0, 0, 0, 0]
#1 from 0x00000902, DLC 8, data [0      0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #1, type 
#2 from 0x00000902, DLC 8, data [0      0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #2, type
FWC: [NOTICE] : CanBus requestRealtimeData
FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485721 DLC 7, [26, 0, 50, 0, 0, 135, 195, 0]
FWC: [NOTICE] : CanBus requestBalanceData
FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485721 DLC 3, [26, 0, 79, 0, 0, 0, 0, 0]
#1 from 0x00000902, DLC 8, data [0      0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #1, type
#2 from 0x00000902, DLC 8, data [0      0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #2, type 
#3 from 0x00000902, DLC 8, data [0      0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #3, type
FWC: [NOTICE] : CanBus requestRealtimeData
FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485721 DLC 7, [26, 0, 50, 0, 0, 135, 195, 0]
FWC: [NOTICE] : CanBus requestBalanceData
FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485721 DLC 3, [26, 0, 79, 0, 0, 0, 0, 0]
#1 from 0x00000902, DLC 8, data [0      0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #1, type
#2 from 0x00000902, DLC 8, data [0      0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #2, type
FWC: [NOTICE] : CanBus requestRealtimeData
FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485721 DLC 7, [26, 0, 50, 0, 0, 135, 195, 0]
FWC: [NOTICE] : CanBus requestBalanceData
FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485721 DLC 3, [26, 0, 79, 0, 0, 0, 0, 0]
```

thankthemaker, 2022-04-18:
`FWC: [ERROR] CANBUS initialization failed
`
Did you check the VESC-ID in VESC-Tool matches the one configured in rESCue? rESCue assumes ID 25 for VESC by default and uses 26 (rESCue itself) and 27 (BLE-proxy) for. To be precise, it uses VESC-ID+1 and VESC-ID+2 for its own IDs. 

From the Log I'd say, the initialization failed because the IDs didn't match.

Koxx3, 2022-04-18:
it was my first mistake. I've just changed it to "1" (my first VESC core ID is 1 and the second is 2).
I am now trying without traces (I noticed it can have an impact)

Koxx3, 2022-04-18:
reading MC config : working
reading APP config : working
writing MC config : NOT working
writing APP config : NOT working

Koxx3, 2022-04-18:
> `FWC: [ERROR] CANBUS initialization failed ` Did you check the VESC-ID in VESC-Tool matches the one configured in rESCue? rESCue assumes ID 25 for VESC by default and uses 26 (rESCue itself) and 27 [...]

you mean I can't use it with 2 VESCs ? or I need to set only the higher ID (2 in my case), so it will use 3&4 ? it will still manage to access 1 ?

thankthemaker, 2022-04-18:
rESCue doesn't support multiple VESC at the moment. I haven't dealt with multiple VESCs so I don't have any experience.

Koxx3, 2022-04-18:
> rESCue doesn't support multiple VESC at the moment. I haven't dealt with multiple VESCs so I don't have any experience.

ahhhhhh ! ok ! it eaplains some things then ;)
I'll cut CAN between both ESC cores to test with a single ESC.

Koxx3, 2022-04-18:
same issue while trying to write MC configuration with single VESC (VESC CAN ID = 1 / config.h = 1) :

```
#4 from 0x00000502, DLC 8, data [21     0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #4, type fill rx buffer
#5 from 0x00000502, DLC 8, data [28     0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #5, type fill rx buffer
#6 from 0x00000502, DLC 7, data [35     0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #6, type fill rx buffer
#7 from 0x00000702, DLC 6, data [1      1       0       41      112     202     0       0]
FWC: [VERBOSE] : CanBus processed frame #7, type process rx buffer for COMM_GET_DECODED_BALANCE
FWC: [NOTICE] : CanBus requestRealtimeData
FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485697 DLC 7, [2, 0, 50, 0, 0, 135, 195, 0]
#1 from 0x00000502, DLC 8, data [0      50      0       0       135     195     1       64]
FWC: [VERBOSE] : CanBus processed frame #1, type fill rx buffer
#2 from 0x00000502, DLC 8, data [7      252     25      0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #2, type fill rx buffer
#3 from 0x00000502, DLC 8, data [14     0       2       33      0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #3, type fill rx buffer
#4 from 0x00000502, DLC 6, data [21     0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #4, type fill rx buffer
#5 from 0x00000702, DLC 6, data [1      1       0       26      54      145     0       0]
FWC: [VERBOSE] : CanBus processed frame #5, type process rx buffer for COMM_GET_VALUES_SELECTIVE
FWC: [NOTICE] : CanBus requestBalanceData
FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485697 DLC 3, [2, 0, 79, 0, 0, 0, 0, 0]
#1 from 0x00000502, DLC 8, data [0      79      0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #1, type fill rx buffer
#2 from 0x00000502, DLC 8, data [7      0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #2, type fill rx buffer
#3 from 0x00000502, DLC 8, data [14     0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #3, type fill rx buffer
#4 from 0x00000502, DLC 8, data [21     0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #4, type fill rx buffer
#5 from 0x00000502, DLC 8, data [28     0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #5, type fill rx buffer
#6 from 0x00000502, DLC 7, data [35     0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #6, type fill rx buffer
#7 from 0x00000702, DLC 6, data [1      1       0       41      112     202     0       0]
FWC: [VERBOSE] : CanBus processed frame #7, type process rx buffer for COMM_GET_DECODED_BALANCE
FWC: [VERBOSE] : CanBus name=UBOX_V2_75_3, dutycycle=0.000000, erpm=0.000000, current=0.000000, ampHours=0.000000, ampHoursCharged=0.000000, wattHours=0.000000, wattHoursCharged=0.000000, mosfetTemp=32.000000, motorTemp=-99.900000, inputVoltage=54.500000, tachometer=0.000000, pidOutput=0.000000, pitch=0.000000, roll=0.000000, loopTime=0, motorCurrent=0.000000, motorPosition=0.000000, balanceState=0, switchState=0, adc1=0.000000, adc2=0.000000, fault=0
FWC: [NOTICE] : CanBus requestRealtimeData
FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485697 DLC 7, [2, 0, 50, 0, 0, 135, 195, 0]
#1 from 0x00000502, DLC 8, data [0      50      0       0       135     195     1       64]
FWC: [VERBOSE] : CanBus processed frame #1, type fill rx buffer
#2 from 0x00000502, DLC 8, data [7      252     24      0       1       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #2, type fill rx buffer
#3 from 0x00000502, DLC 8, data [14     0       2       33      0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #3, type fill rx buffer
#4 from 0x00000502, DLC 6, data [21     0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #4, type fill rx buffer
#5 from 0x00000702, DLC 6, data [1      1       0       26      110     132     0       0]
FWC: [VERBOSE] : CanBus processed frame #5, type process rx buffer for COMM_GET_VALUES_SELECTIVE
FWC: [NOTICE] : CanBus requestBalanceData
FWC: [VERBOSE] : CanDevice Sending CAN frame 2147485697 DLC 3, [2, 0, 79, 0, 0, 0, 0, 0]
#1 from 0x00000502, DLC 8, data [0      79      0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #1, type fill rx buffer
#2 from 0x00000502, DLC 8, data [7      0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #2, type fill rx buffer
#3 from 0x00000502, DLC 8, data [14     0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #3, type fill rx buffer
#4 from 0x00000502, DLC 8, data [21     0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #4, type fill rx buffer
#5 from 0x00000502, DLC 8, data [28     0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #5, type fill rx buffer
#6 from 0x00000502, DLC 7, data [35     0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #6, type fill rx buffer
[... cut ...]
FWC: [VERBOSE] : CanBus processed frame #1, type fill rx buffer
#2 from 0x00000502, DLC 8, data [7      0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #2, type fill rx buffer
#3 from 0x00000502, DLC 8, data [14     0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #3, type fill rx buffer
#4 from 0x00000502, DLC 8, data [21     0       0       0       0       0       0       0]
FWC: [VERBOSE] : CanBus processed frame #4, type fill rx buffer
```

thankthemaker, 2022-04-18:
Hmm, from the looks it looks like writing is ok. The VESC sends back the current config after writing and it looks like packages get in a mess. I guess you have enabled VESC RT log or RT data and packages don't arrive in the correct order. I have to have a look at it and maybe manually sort packages.

Could you please check it the config really wasn't updated? Could you please also test writing with RT Data deactivated in VESC?

Koxx3, 2022-04-18:
writting seems good, but datas are not really written in the VESC. read erase data I set in vesctool.
RT & RT app aren't enabled on vesctool (USB disconnected & not on the realtime page of the VESCTool mobile app).
I'll disable balance & realtime data reading from rescue to have a better view.

thankthemaker, 2022-04-18:
Which versions (VESC Tool and VESC firmware) are you using? Maybe I can test by myself.

Foujiwara, 2022-04-18:
Great project, we love it thxthemaker thx you for it, please help us to make dual VESC work on the last revision of VESC firmware =)

Koxx3, 2022-04-18:
> Which versions (VESC Tool and VESC firmware) are you using? Maybe I can test by myself.

FW 5.3 / Vesctool 3.01

Foujiwara, 2022-04-18:
in my case i use V3.01 Vesctool app and V6 bêta vesc firmware, but Koxx is on 5.3 vesc firm

Koxx3, 2022-04-18:
new logs with only interesting things : 
[rescue1.txt](https://github.com/thankthemaker/rESCue/files/8505210/rescue1.txt)

Koxx3, 2022-04-18:
@thankthemaker : if you have telegram, I think if would be much easier to discuss ;)

Koxx3, 2022-04-18:
with BLE data displayed :
[rescue2.txt](https://github.com/thankthemaker/rESCue/files/8505406/rescue2.txt)

Koxx3, 2022-04-18:
I think BLE is overwelmed when writing from mobile to ESP32.
Frame seems incomplete (no 03 at the end)

Note that my Samsung phone as a MTU of 20 for those BLE writings

```
14:01:07.920 > FWC: [NOTICE] : BleServer onWrite to characteristics: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
14:01:07.922 > FWC: [NOTICE] : BleServer onWrite - value: 020, data:  03 01 dd 10 de 88 e0 57 01 00 00 03 e8 00 00 00 00 00 00 32
14:01:07.923 > FWC: [VERBOSE] : BleCanProxy Proxy in, command 16, length 477
14:01:07.924 >
14:01:07.924 > FWC: [NOTICE] : BleServer onWrite to characteristics: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
14:01:07.925 > FWC: [NOTICE] : BleServer onWrite - value: 020, data:  02 00 01 07 00 00 00 47 43 50 00 00 00 03 00 46 6a 60 00 3e
14:01:07.927 > FWC: [NOTICE] : BleServer onWrite to characteristics: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
14:01:07.928 > FWC: [NOTICE] : BleServer onWrite - value: 020, data:  19 99 9a 3f 80 00 00 40 00 00 00 3f c0 00 00 01 01 00 00 00
14:01:07.929 > FWC: [NOTICE] : BleServer onWrite to characteristics: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
14:01:07.930 > FWC: [NOTICE] : BleServer onWrite - value: 020, data:  00 00 00 00 00 02 3e cc cc cd 3e 4c cc cd 01 00 45 3b 80 00
14:01:07.931 > FWC: [NOTICE] : BleServer onWrite to characteristics: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
14:01:07.932 > FWC: [NOTICE] : BleServer onWrite - value: 020, data:  0f a0 3d 8f 5c 29 40 40 00 00 00 3e 19 99 9a 3f 66 66 66 40
14:01:07.933 > FWC: [NOTICE] : BleServer onWrite to characteristics: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
14:01:07.934 > FWC: [NOTICE] : BleServer onWrite - value: 020, data:  40 00 00 40 00 00 00 3f 66 66 66 40 40 00 00 01 01 00 00 00
14:01:07.936 > FWC: [NOTICE] : BleServer onWrite to characteristics: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
14:01:07.937 > FWC: [NOTICE] : BleServer onWrite - value: 020, data:  00 00 00 00 00 00 00 00 00 02 3e 99 99 9a 3d cc cc cd 01 00
14:01:07.938 > FWC: [NOTICE] : BleServer onWrite to characteristics: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
14:01:07.939 > FWC: [NOTICE] : BleServer onWrite - value: 020, data:  45 3b 80 00 01 f4 00 01 c2 00 01 3e 19 99 9a 3e cc cc cd 3e
14:01:07.940 > FWC: [NOTICE] : BleServer onWrite to characteristics: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
14:01:07.941 > FWC: [NOTICE] : BleServer onWrite - value: 020, data:  4c cc cd 45 3b 80 00 00 00 00 00 00 00 00 00 02 00 01 45 3b
14:01:07.943 > FWC: [NOTICE] : BleServer onWrite to characteristics: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
14:01:07.944 > FWC: [NOTICE] : BleServer onWrite - value: 020, data:  80 00 00 3d 8f 5c 29 40 40 00 00 01 03 01 00 03 4c c6 c7 00
14:01:07.945 > FWC: [NOTICE] : BleServer onWrite to characteristics: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
14:01:07.946 > FWC: [NOTICE] : BleServer onWrite - value: 020, data:  01 00 00 00 00 00 00 00 00 00 00 00 00 03 e8 00 00 41 a0 00
14:01:07.947 > FWC: [NOTICE] : BleServer onWrite to characteristics: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
14:01:07.948 > FWC: [NOTICE] : BleServer onWrite - value: 020, data:  00 42 34 00 00 3f 66 66 66 00 00 00 00 00 00 00 00 00 00 00
14:01:07.949 > FWC: [NOTICE] : BleServer onWrite to characteristics: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
14:01:07.950 > FWC: [NOTICE] : BleServer onWrite - value: 020, data:  00 00 00 00 00 00 00 03 e8 03 e8 01 2c 02 ee 03 e8 01 2c 43
14:01:07.952 > FWC: [NOTICE] : BleServer onWrite to characteristics: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
14:01:07.953 > FWC: [NOTICE] : BleServer onWrite - value: 020, data:  48 00 00 03 e8 01 2c 00 00 00 00 00 c8 00 00 00 00 01 f4 00
14:01:07.954 > FWC: [NOTICE] : BleServer onWrite to characteristics: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
14:01:07.955 > FWC: [NOTICE] : BleServer onWrite - value: 020, data:  00 00 00 00 00 00 00 01 2c 41 a0 00 00 41 00 00 00 41 f0 00
14:01:07.956 > FWC: [NOTICE] : BleServer onWrite to characteristics: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
14:01:07.957 > FWC: [NOTICE] : BleServer onWrite - value: 020, data:  00 00 00 00 00 00 00 00 00 00 00 0a 00 00 00 00 00 00 00 00
14:01:07.958 > FWC: [NOTICE] : BleServer onWrite to characteristics: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
14:01:07.959 > FWC: [NOTICE] : BleServer onWrite - value: 020, data:  40 00 00 00 00 00 00 00 40 a0 00 00 3f 80 00 00 00 64 40 a0
14:01:07.961 > FWC: [NOTICE] : BleServer onWrite to characteristics: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
14:01:07.962 > FWC: [NOTICE] : BleServer onWrite - value: 020, data:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

Koxx3, 2022-04-18:
More interesting logs...

after changing MTU to 120 (instead of 512, too much for my phone and not negociated ... BLE MTU negociation works pretty bad by the way) and increasing Nimble buffers : 
- read FW : ok
- read APP conf : ok
- read MC conf : ok
- wite APP conf : KO
- wite MC conf : ok

note how the write app doesn't continue. It's weird. Datas are written in the VESC this time. but VESC => BLE feedback doesn't work (see rescue_write_app_conf.txt).

[rescue_write_app_conf.txt](https://github.com/thankthemaker/rESCue/files/8505594/rescue_write_app_conf.txt)
[rescue_write_mc_conf.txt](https://github.com/thankthemaker/rESCue/files/8505595/rescue_write_mc_conf.txt)
[rescue_read_app_conf.txt](https://github.com/thankthemaker/rESCue/files/8505596/rescue_read_app_conf.txt)
[rescue_read_mc_conf.txt](https://github.com/thankthemaker/rESCue/files/8505597/rescue_read_mc_conf.txt)

thankthemaker, 2022-04-18:
> new logs with only interesting things : [rescue1.txt](https://github.com/thankthemaker/rESCue/files/8505210/rescue1.txt)

Sorry, I'm out with the family at the moment. I'll be back in the evening. I'll have a look at you logs and your comments once im back at home. I'll also send you my telegram number.

thankthemaker, 2022-04-18:
Ok, I never had a problem with the BLE connection. Neither with my macbook nor with a Huawei Mate. But it makes sense.
Both write operations look good so far and normally the VESC should answer with a read command of the same config data. For the motor config it seems like it worked but for the app config the read command is missing. 
At the moment I don't have an idea why this happened.

thankthemaker, 2022-04-18:
You can also find me on Telegram...
https://t.me/thankthemaker