thankthemaker/rESCue #35: Bug? after idle time, no front and back litgh restart.

MadeInFondDuGarage, 2022-04-29:
when i use it, i wait some long time, (discus with my frien).
front and back ligth go off (idle time).
i put my foot on, and nothing change, the back and front alway off, and the litghbar dont go to blue. the board steel ok i can ride.

i need a restart (switch off /on) to recover the litgh.

thankthemaker, 2022-05-11:
Could you please tell me which firmware version you're on?
Does this happen every time your rESCue enters idle mode or only sometimes? I wasn't able to reproduce so far.

MadeInFondDuGarage, 2022-05-14:
hello i make more test, with my son to see the ligth.
after this, i can t  reproduce it.
what i see.
i turn on the board, i have the start light, and the idle. if i put the max time after the time light go off, if i put zero light alway on.
i go ride, the front and back litgh are ok.
i stop , the light directly go off (no idle light)
i restart the ride, the ligth go on but not imediatly, i need to go a little to active them.

thankthemaker, 2022-10-09:
Ok, today I tried to reproduce with version 2.2.1 but I had no luck. 
Could you please tell me which version your on and maybe try with 2.2.1 again, if you're running another version. 
Could you please also check if "verbose" logging is enabled in rESCue app and what "Can status rate" and "Can status message mode" you've set in VESC tool.

I guess it's a combination of MCU overload due to verbose logging enabled and a high  "status rate". You could try with 
only 5Hz and CAN_STATUS_1 and Loglevel "notice" in rESCue App.

If the issue still occurs, please reopen the issue

closed 2022-10-09 by thankthemaker