thankthemaker/rESCue #42: Bug: Battery Indicator Footpad Indication not adjusting

ty8447, 2022-07-29:
When half of the footpad is stepped on, the amount of LEDs that turn on to show what footpad is pressed does not adjust depending on the size of the strip used. When an even number of LEDs is specified it should divide the strip in half and use that, and when an odd number of LEDs is specified it should subtract 1, divide the number of LEDs in half and then use that amount.

What would make a lot of sense would be to add an option in the settings menu on the app to customize how many LEDs are used for that detection and the padding in the middle of the two halves.

thankthemaker, 2022-10-09:
Thank's for your feedback. The number of pixels is already adjustable in the wizard and also in the rESCue-App settings, it is still called "Battery Monitor -> Number of pixel". I'll change this with the next update to somethong more meaningful.

But you're still right regarding the odd/even problem.  I'll fix it in the next firmware update

thankthemaker, 2022-10-09:
Fixed in 2.2.2

closed 2022-10-09 by thankthemaker