thankthemaker/rESCue #57: Can not get BLE-Uart bridge to work

aka13-404, 2022-12-11:
Great coneption! Sadly, seemingly my incompetence  does not permit me to get the BLE bridge working.
I flash the esp successfully, I am connected to the uart on the VESC (I had an nrf working properly there), I am soldered to GPIO 16 &17, I have APP set to UARD on my single ubox. I can successfully connect to the flashed ESP, but get "make sure port belongs to vesc". 
What could I be doing wrong?
[image]

thankthemaker, 2022-12-11:
How did you build the rEScue firmware? It's preconfigured to use CANBUS, to switch to UART you have to build it with CANBUS-only disabled in config.h

Also make sure to connect the ESP and the VESC to the same GND, UART needs a common GND.

aka13-404, 2022-12-11:
I have to admit, this is the very first time I build anything for embed myself.
I never bothered to check any configs, since the readme did not mention them.

What would I configure, if the only thing I need is the uart-ble bridge?

aka13-404, 2022-12-11:
[image]
Well, this has worked perfectly. I connect and read VESC as expected.
I tried to comment out all the LED stuff (for now it is unlikely, that I will run LEDs), but it errored out. Guess, there is no "no led mode" :D 
Thank you again, you saved my evening.

closed 2022-12-11 by aka13-404

thankthemaker, 2022-12-11:
Sorry, I mismatched CANBUS_Enabled and CANBUS_only.
Why did you need to comment out LED stuff?

aka13-404, 2022-12-11:
Well, as mentioned, right now and for the foreseeable future I only care about the ble-uart bridge. All else looks pretty sweet, but right now I don't use it, so I thought that perhaps it's modular and I should disable it for now.

svenkvade, 2023-02-06:
If I were to only use this as a replacement for the bluetooth modules as a BLE-UART bridge, would this run on an ESP8266 as well? Or better to just get a 32 regardless? Thanks

topoll113, 2023-02-07:
@thankthemaker same as author Im not able to make UART work with VESC tool . I out commented " #define CANBUS ENABLED " in config.h and also changed " default_envs = wemos_d1_mini32 " to " default_envs = nodemcu-32s " in platformio.ini in order to match my board. rESCue app reading VESC values . I hooked ESP32 with VESC 75100. I tried 6.0, 5.02 and 5.01 firmware but no luck with any of them... Any idea ?

thankthemaker, 2023-02-08:
> @thankthemaker same as author Im not able to make UART work with VESC tool . I out commented " #define CANBUS ENABLED " in config.h and also changed " default_envs = wemos_d1_mini32 " to " default_envs = nodemcu-32s " in platformio.ini in order to match my board. rESCue app reading VESC values . I hooked ESP32 with VESC 75100. I tried 6.0, 5.02 and 5.01 firmware but no luck with any of them... Any idea ? Edit: rESCue not working with VESC Tool 6.0 but It does with 3.01. Any chance that this could be fixed ?

It should work, have yuo checked the PINs? The ESP32 has 3 UART ports, which one did you use?
It's also important to use a common GND for ESP32 and VESC.

topoll113, 2023-02-08:
> It should work, have yuo checked the PINs? The ESP32 has 3 UART ports, which one did you use? It's also important to use a common GND for ESP32 and VESC.

Wiring is okay. I connected UART to pin 16 & 17. I’m getting reading in rESCue app and I’m able to connect with VESC Tool ver. 3.01. When I’m trying connect with VESC Tool ver. 6 I’m getting message: “ Could not read firmware version. Make sure that selected port really belongs to VESC”.