thankthemaker/rESCue #61: Connecting a ESP32 devkit v1 to Flipsky fsesc75100

LankoG, 2023-07-11:
I am trying to connect esp32 with a flipsky vesc, but cannot get esp to work. I connected it only through UART because I do not have CAN bus transceiver.

thankthemaker, 2023-07-11:
For UART to work you'll have to rebuild the firmware with CANBUS_ONLY commented out in config.h

Otherwise CANBUS is used.

LankoG, 2023-07-11:
I already did that, also tried to change [platformio] default_envs = nodemcu-32s, but nothing worked. I am using vs code to upload the code with platformio, but don't think there is anything wrong there.

thankthemaker, 2023-07-11:
Ok, did you check that UART is enabled in VESC tool?
Do your VESC and ESP share a common ground?
Using a common ground is important.

LankoG, 2023-07-11:
yes when I used just the VescUart library it was able to print the data to the terminal

LankoG, 2023-07-11:
I tried reinstalling the source code and now I get this error:
[image]
[image]

barsikus007, 2023-12-09:
You also need to disable CANBUS_ENABLED too

btw after uploading, issue still presist

jwidess, 2024-07-24:
Oh my god thank you for this thread! I've spent the last two days trying to figure out why UART wasn't working properly. The fact that **CANBUS_ONLY** is the default **NEEDS** to be mentioned somewhere!

NOTE: The "**-D CANBUS_ENABLED -D CANBUS_ONLY**" flags are located in the Environment Variables in "**platformio.ini**" **not** in config.h. To **enable UART** on my nodemcu32, on lines 141-152 of `platformio.ini` I removed the CANBUS flags so my config looks like the following. 

```ini
[env:nodemcu-32s]
platform = espressif32
board = nodemcu-32s
framework = arduino
monitor_speed = 115200
board_build.partitions = default.csv
lib_deps =
    ${common_env_data.lib_deps_external}
build_flags =
     ${ESP32.build_flags} -D LED_WS28xx
extra_scripts = pre:change_firmware_name.py
custom_firmware_name = firmware_nodemcu-32s
``` 

**Errors I was having:**
**rESCue App:** Always in the Disconnected State
**VESC-Tool:** "Could not read firmware version. Make sure that the selected port really belongs to the VESC"

commit 457a7b25 (2025-03-15): Merge pull request #96 from wildekek/master

Support for esp32doit_devkit_v1(_uart) fixes #61
commit 96f2ab1d (2025-03-15): Support for esp32doit_devkit_v1(_uart) fixes #61
closed 2025-03-15 by thankthemaker