thankthemaker/rESCue #63: German brakelights

GulliUlli, 2023-08-01:
Hi. So i saw, that it gives brakelight Support. Is it possible to make it usable in Germany with a static breaklight (Like Cars) instead of flashing light?

GulliUlli, 2023-08-01:
Ok i Unterstand. So can i Change it by myself then? That i have 60% backlight And 100% Static brakelight? It is it fixed to flashing in the code now?

thankthemaker, 2023-08-02:
With the current firmware, the brake light brightness is fix at maximum. It's possible to change the brightness of the riding lights. In the default settings the brake lights are set to maximum (255) and the riding lights to 230.

It's also possible to completely switch off the brake lights and to change the value of the regenerative inrush current (default 4A) which triggers the brake lights.

If you want to get an overview of the settings, go and get the App for free and use the virtual rESCue device (change in settings after download) to look around and play with all options:

ios: https://apps.apple.com/de/app/rescue-app/id1572076731
android: https://play.google.com/store/apps/details?id=org.thankthemaker.rescueapp

GulliUlli, 2023-08-02:
Okey i will try it. Thx 🙏

closed 2026-11-02 by thankthemaker