thankthemaker/rESCue #64: BLE Brigde Not working with new Vesc Tool App

SyedShahzebHussain, 2023-08-21:
I am trying to connect Vesc tool app V6.05, as soon as i connect ble, ESP32 gets reset and if i connect vesc tool V1.25 app everything works fine. can you help me with this issue. here is the error when i connect with new vesc tool

rst:0xc (sw_cpu_reset),boot:0x13 (spi_fast_flash_boot) configsip: 0, spiwp:0xee clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00 mode:dio, clock div:1 load:0x3fff0018,len:4 load:0x3fff001c,len:1044 load:0x40078000,len:10124 load:0x40080400,len:5856 entry 0x400806a8

Starcult, 2023-08-21:
I am using vesc tool v6.02 (iphone) no problem at all .
Are you connect it from uart or can?

SyedShahzebHussain, 2023-08-22:
@Starcult I am using Android Phone and connected it from UART, as soon i connect to bluetooth using 6.05 Version App, Esp32 resets.

closed 2023-08-23 by SyedShahzebHussain