thankthemaker/rESCue #88: lightbar footpad sensor VISUAL indications not working on 2.4.1 up to 2.5.4

guerradr, 2023-12-06:
Hardware-Version: v3.1 (old board biggest one)

the lightbar footpad sensor VISUAL indications are not working on firmware versions 2.4.1 and up to 2.5.4

I reverted back to 2.3.3 and then they work.

thankthemaker, 2023-12-10:
Could you please give me some details? What isn't working?
I just tested 2.5.4 with my board and indicators seem to work.

https://github.com/thankthemaker/rESCue/assets/33477197/cae385dd-faef-42c1-b84c-a188a6ac5178

guerradr, 2023-12-10:
Yeah its wierd. The startup sequence and battery indication work fine. But the footpad sensor indicatior did not show on the lightbar. Only when I downgraded to the stated version nr. did it work again for me. Do you need further info?

https://github.com/thankthemaker/rESCue/assets/33477197/cae385dd-faef-42c1-b84c-a188a6ac5178

guerradr, 2024-01-09:
Here is a short video. Apparently it's hard to replicate the behaviour when switching firmware versions. Could it be that the eeprom settings are somehow retained when flashing new firmware?
https://youtu.be/6uUoVD5nSK8?si=3yO6029yzBX9r0gn

thankthemaker, 2024-01-11:
Thanks for the video, it makes it much clearer. I'll check it again in the evening. Looks like pushing the left hand side gets recognized like an ADC-FULL while the right hand side is recognized as ADC-HALF, which is correct if only one side is pressed.

guerradr, 2024-01-11:
Yes that seems so. I’m on the newest VESC 6.02 and I have the same behavior on the UBOX SINGLE 100V and the Raiden7. I have them wired up via CAN-BUS.

thankthemaker, 2024-01-16:
Did you find some time to check the values in VESC tool?

I tried to reproduce the behavior but wasn't successful. I still think something is wrong with your sensor / wiring.

guerradr, 2024-01-16:
Hi, the settings are fine in the VESC Tool. You can also see in the video that the board only activates when pressing both footpad sides down. I see the same behaviour on both of my boards (UBOX Single with rESCue light and Raiden 7 with rESCue s3). There is not much I can do wrong with my wiring since I'm can bus only and i have connection via VESC Phone App. Since I had different behaviours and indications when swapping between rESCue Software Versions - are there any setting values retained in the eeprom when flashing?
Maybe you can answer me another question not related to the lightbar. When I want to do the "Setup Motors FOC" via the rESCue bluetooth bridge (CAN-Bus only), I always get an error that it is unable to write the motor settings. However I can use the VESC App fine when changing settings in the  motor and app configs. I always have to use an USB Cable for the initial motor config.

thankthemaker, 2024-01-16:
Sorry for asking again. Could yous check which state is shown in VESC tool when pressing the pads? 
There's nothing configured in rESCue itself, it just checks the state it gets from VESC. So it would be very helpful to know what the stat looks like in VESC.

guerradr, 2024-01-16:
I meant the footpad indicator configuration. When I use the rESCue App I sometimes had to save the config two or three times to have the whole configuration saved on the esp32. Thus seeing different behaviours between the saved configs. I used different phones to check if my app had a problem.
Here is a screen record of my VESC App showing correct behaviour of my footpads.
https://youtu.be/eR7gBGYLcno

thankthemaker, 2024-01-17:
Thanks for the video. You're right. The behaviour in VESC tool looks as it should.
I have an idea what's going wrong, but I first have to try to reproduce the problem to fix the issue.
Thanks for your help!

0x41kravchenko, 2024-06-25:
Hi guys! I had exactly same situation and after checking the logs I saw "Unknown switch state" log. The cause was that in vescData.switchState the value was 3 for fully activated footpads. After logging switchState and comparing output from VESC tool I got following relations:
- switchState==0 - when adc1 and adc2 are both 0
- switchState==1 - when adc1 is active
- switchState==2 - when adc2 is active
- switchState==3 - when adc1 and adc2 are both active
Basically now (in relatively new VESC firmwares?) switchState fully corresponds to AdcState enum values, following code change confirmed it:
[image]

But I'm not sure about old versions of VESC firmwares and whether 3rd value of switchState was added recently.
I have 6.2 CheapFOCer2.