thankthemaker/rESCue #89: 0 pins division by zero

barsikus007, 2023-12-10:
When I setup rescue app I specified 0 pins (cause I have no leds at all, I want to use it as BLEproxy rn)

Now I have bootloop with this
```log
09:51:29.384 > ets Jun  8 2016 00:22:57
09:51:29.387 >
09:51:29.387 > rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
09:51:29.393 > configsip: 0, SPIWP:0xee
09:51:29.393 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
09:51:29.401 > mode:DIO, clock div:2
09:51:29.401 > load:0x3fff0030,len:1184
09:51:29.404 > load:0x40078000,len:13232
09:51:29.407 > load:0x40080400,len:3028
09:51:29.411 > entry 0x400805e4
09:51:30.672 > Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.
09:51:30.678 >
09:51:30.678 > Core  1 register dump:
09:51:30.680 > PC      : 0x400d6b97  PS      : 0x00060a30  A0      : 0x800d736f  A1      : 0x3ffcb4c0  
09:51:30.689 > A2      : 0x3ffbbfc8  A3      : 0x3ffc38c4  A4      : 0x000000ff  A5      : 0x00000000
09:51:30.697 > A6      : 0x3f400547  A7      : 0x3ffc3860  A8      : 0x00630000  A9      : 0x00000064  
09:51:30.706 > A10     : 0x00000000  A11     : 0x00000000  A12     : 0x000003e8  A13     : 0x00000043
09:51:30.711 > A14     : 0x7ff00000  A15     : 0x7ff0cccc  SAR     : 0x00000020  EXCCAUSE: 0x00000006  
09:51:30.719 > EXCVADDR: 0x00000000  LBEG    : 0x40085c39  LEND    : 0x40085c41  LCOUNT  : 0x00000027
09:51:30.728 >
09:51:30.728 >
09:51:30.728 > Backtrace: 0x400d6b94:0x3ffcb4c0 0x400d736c:0x3ffcb4e0 0x400ea2e6:0x3ffcb510
09:51:30.736 >
09:51:30.736 >
09:51:30.736 >
09:51:30.736 >
09:51:30.736 > ELF file SHA256: 0a949e50ce96eca6
09:51:30.738 >
09:51:30.889 > Rebooting...
```
repeating at serial
[image]
this is stacktrace decoding

barsikus007, 2023-12-10:
and can you provide me any way to reset this, please

thankthemaker, 2023-12-10:
Ok, I see this is indeed a bug. I will fix this bug by deactivating lights if pixel count is set to zero, but for the moment I guess you could work around by simply setting pixel count > 0. Without a LED strip connected, the code will still work.

barsikus007, 2023-12-10:
I can't change settings due to bootloop

btw I changed code to reset config and now all works properly

thankthemaker, 2023-12-10:
Ok, glad you found a solution.You could also use the webinstaller and erase flash during installation.https://rescue.thank-the-maker.org/install