thankthemaker/rESCue #95: Link to app

wildekek, 2025-03-12:
Thanks for a great project!

I could not find the rESCue app in the iOS app store. After some googling I found you have a [Web Bluetooth API app](https://rescue.thank-the-maker.org/device), which works great!

I'd love to update the README for you and create a PR. Are the iOS/Android apps still in their respective stores and do you have the url? I'd also like to add the web app to the docs if you don't mind.

thankthemaker, 2025-03-13:
Seems like the Apps have been removed from the AppStore / Playstore as they haven't been updated for quite a while.
I have to check of my developer profiles are still active and I can upload them for review again.

thankthemaker, 2025-03-15:
The iOS App is available at:
https://apps.apple.com/us/app/rescue-app/id1572076731